3DCP-TUe/SaladSlicer #79: Salad Slicer does not work on Rhino 7 Mac

highnze, 2023-10-27:
Hello :) 

I put the file "SaladSlicer.Gh.gha" into the "Libraries" Folder (Components Folder). When I start grasshopper, there is this Error showing up. When I start the example file "01_Planar_2D_Slicer.gh" the window "Unrecognized Objects" is coming up. (see Image) 

Is this a problem with Rhino for Mac? 

Best regards
Lisa 

[image]

arjendeetman, 2023-10-27:
Hi Lisa,

We didn't test it on Mac. I also don't have experience with it. Based on your information, did you also place the `SaladSlicer.dll` file in the libraries folder? It looks like you are missing this file. It probably includes the missing objects. 

I don't know if this will fix the issue since I don't know what the requirements are for plugins for Mac.

Best,

Arjen

arjendeetman, 2023-11-24:
Hi @highnze,

Could you fix this issue? If it runs on Mac I can also make it available for this OS on the package manager. 

Best,

Arjen